LibreSolar/learn.libre.solar #1: Fix rendering of mathematical equations

martinjaeger, 2019-06-12:
The mathjax plugin is already installed as [described in VuePress documentation](https://vuepress.github.io/en/plugins/mathjax/):
https://github.com/LibreSolar/learn.libre.solar/blob/086e2c88ac7ad2c597a55c702ad1b699dfd8cdc3/docs/.vuepress/config.js#L6-L11

Unfortunately, math formula are not rendered correctly. Example for test:
https://learn.libre.solar/development/dcdc_converter.html

martinjaeger, 2019-06-12:
@jurra do you have an idea what could be the problem?

jurra, 2019-06-12:
Hi Martin I will check it when i get home. Today i am working. I havent
work with this module. It looks like some config issue.

jurra, 2019-06-12:
I solved this issue by updating vuepress and changing the package to vuepress-plugin-latex. I tried to push my changes to my own branch, but I don't have rights to push the changes:
[image]

martinjaeger, 2019-06-12:
Very cool!

It's strange that you can't push. Checked again, you have write (incl. push) permission. Or otherwise try with a pull-request.

jurra, 2019-06-12:
No I pushed, its ok ;)

martinjaeger, 2019-06-14:
Solved after upgrade of VuePress to v1.0.

closed 2019-06-14 by martinjaeger