LibreSolar/learn.libre.solar #2: Embed tool for interactive plotting

martinjaeger, 2019-06-12:
Suggestions:
- [Dygraphs](http://dygraphs.com/index.html): Using javascript only
- Plot.ly: Python necessary?
- [D3](https://d3js.org/): Also pure javascript

jurra, 2019-06-12:
It might be a good idea to use `vuejs` components instead of direct javascript.

martinjaeger, 2019-06-13:
Ok, didn't know that this is possible directly in `vuejs`.

The idea: The user should be able to change some parameters in text fields. After that, a graph should be automatically updated. This could be very useful for the system layout section, e.g. to see the temperature influence on the performance of solar panels by changing the parameters and getting direct results.

This example goes in a similar direction, but I don't like the style of dygraphs a lot: http://dygraphs.com/gallery/#g/plotter

There seems to be an infinite amount of javascript graphing libraries, lots of them can somehow be integrated in vue.js, as shown [here](https://madewithvuejs.com/charts). Do you have a recommendation where to start? Have you used one specific graphing library already?

jurra, 2019-06-13:
yes :)
1. Point to a graph example that you want to make. Maybe then we can change the parameters, equations, etc.
2. Then I can create a vuejs component.
3. Are there more graphs of this kind you want to reproduce? Maybe it is interesting to make them programatic to a certain extent. I mean that they can be reusable. This is a bit more complicated, so lets try first step 1 and 2.

martinjaeger, 2019-06-13:
Cool :)

Answering point 3 first: Yes, possibly it will be useful to have graphs to play around with and change dynamically in several different parts of the vuepress site, especially in the "Component Development" category. I think it helps a lot to understand influencing factors if it's not just a static graph but something you can change.

The first chapter + graph I have in mind should help to understand how different types of charge controllers and solar panels interact. Basically I'd like to have the graphs in [this document by Victron Energy](https://www.victronenergy.com/upload/documents/White-paper-Which-solar-charge-controller-PWM-or-MPPT.pdf) in an interactive way. We could start with an interactive solar panel characteristic curve.

Input parameters to define the curve at standard test conditions (STC):
- `Vmpp`: Voltage in maximum power point
- `Impp`: Current in maximum power point
- `Voc`: Open circuit voltage
- `Isc`: Short circuit current

Input parameters for temperature correction
- `alpha`: Temperature coefficient of `Isc`
- `beta`: Temperature coefficient of `Voc`
- `T`: Module temperature

These are the typical parameters you find on each spec sheet of solar modules. All other parameters mentioned in the Victron document can be derived from these ones.

The graph itself is an implicit function, so it can't be easily plotted by a simple formula. I suggest to use a lookup table (i.e. array) and scale it according to the input parameters.

[The MPPT sizing calculator](https://www.victronenergy.com/solar-charge-controllers) is another nice example from Victron Energy, which has an interactive graph of the solar yield already (click at the top right corner).

jurra, 2019-06-13:
@martinjaeger I already read the examples you put, but didn't find an equation to plot the graph.
for the first case if you provide me with the graph functions to visualize them that would make things easier. [See this example](http://dygraphs.com/gallery/#g/plotter). 

But I think this enhancement will definetly require of some development effort. It sounds very interesting all the things you mention and what can be achieved.

With regard to interactivty

rbtnx, 2019-06-13:
@jurra Here is an equation to model the graph Martin mentioned:
$V = V_t \cdot \ln\left(\frac{I_{sc} - I + I_{0}}{I_{0}} - IR_{pv} \right )$
Parameters can be approximated with formulas, to make it easy I precalcuated them with the data from the already mentioned [Victron energy document](https://www.victronenergy.com/upload/documents/White-paper-Which-solar-charge-controller-PWM-or-MPPT.pdf):
$V_t = 1.9102$
$I_{sc} = 6.12$
$I_0 = 7.514 \times 10^{-5}$
$R_{pv} = -0.1737$

Then just plot voltage V on x-axis and current I on y-axis. I can also help you to look into the graph capabilities of vue.js.

rbtnx, 2019-06-13:
Would be nice to also plot the maximum power point, so in the end it looks like this:
[image]

martinjaeger, 2019-06-13:
Cool, didn't know that there is such a simple approximation. At least the German Wikipedia stated a recursive function that could only be solved numerically.

rbtnx, 2019-06-13:
The approximation is derived by a numerical solution, but for the most frequently used solar cells (Si, a-Si, CdTe, etc) it's accurately enought (~1%). See [Photovoltaik Engineering by Wagner](https://www.springer.com/de/book/9783662486405)

jurra, 2019-06-13:
@rbtnx what is ln in the formula? I am making now a very simple script in js declaring the variables to start:
```
let vt = 1.9;
let isc = 6.12;
let i0 = Math.pow(7.5, -5);
let Rpv = -0.1237;
let Vt = 1.9102;

let v = Vt * ln(isc - i + i0 / i0 - i * rpv);
```
I think a nice first step is to just do a calculator with input fields and see how that works, then we can go from there to plotting. The first challenge here is to make the equation reactive with javascript and vue.

jurra, 2019-06-13:
@rbtnx and @martinjaeger I figured out how to approach it in more detail. I give an expanatio here and also the things required to implement a first trial. We need certain input fields as martin mentioned:

Input parameters to define the curve at standard test conditions (STC):
- `Vmpp`: Voltage in maximum power point
- `Impp`: Current in maximum power point
- `Voc`: Open circuit voltage
- `Isc`: Short circuit current

Input parameters for temperature correction
- `alpha`: Temperature coefficient of `Isc`
- `beta`: Temperature coefficient of `Voc`
- `T`: Module temperature

For each of these ones I need the correspondent equations that would be the javascript methods we will be using to recalculate, update the model. So basically I would need those equations to map them to each input field in this vue component we are going to make.

martinjaeger, 2019-06-13:
The seven values I mentioned are all input fields. The user has to fill them, but we can prefill them with reasonable values, e.g. from the Victron file.

> let i0 = Math.pow(7.5, -5);

I guess javascript allows this notation aswell: `7.5e-5`

> let ln = ?

`ln` is the natural logarithm, there should be a function in Javascript.

> let i = ?

This is the current, it should be a parameter going from 0 to `Isc`.

> let v = Vt * ln* (isc - i + i0 / i0 - i * rpv);

Don't forget the brackets:
`let v = Vt * ln ((isc - i + i0) / i0 - i * Rpv);`

jurra, 2019-06-13:
[image]
This image shows the component I added to the site. Go to `.vuepress/components/InputGraph.vue` in `graph` branch. To plot the graph we need ranges for the data, I have no idea about this range data set. How should we obtain it via a formula based on some input?

martinjaeger, 2019-06-14:
The range for the data on y axis is `0` to `Isc`, for the x axis it is `0` to `Voc`.

Kathrin will post the formulas to calculate the parameters for above equation from the datasheet values so that we don't need to take the hard-coded values anymore.

jurra, 2019-06-14:
@Martin Jäger <martin@libre.solar> this is an interesting feature for your
OER, but it will take sometime and effort, at least from my side. I spent
the afternoon and night yesterday looking into d3js and it has some
learning curve. I think the commitment it will take will be quite some. So
its good to talk about time estimates and priorities if you want me to help
you.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

rbtnx, 2019-06-17:
So here are the formulas for the approximation: Given `Impp`, `Vmpp`, `Isc` and `Voc` you can calculate `V` with

<a href="https://www.codecogs.com/eqnedit.php?latex=V&space;=&space;V_t&space;\cdot&space;\ln\left(\frac{I_{sc}&space;-&space;I&space;&plus;&space;I_{0}}{I_{0}}&space;-&space;IR_{pv}&space;\right&space;)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?V&space;=&space;V_t&space;\cdot&space;\ln\left(\frac{I_{sc}&space;-&space;I&space;&plus;&space;I_{0}}{I_{0}}&space;-&space;IR_{pv}&space;\right&space;)" title="V = V_t \cdot \ln\left(\frac{I_{sc} - I + I_{0}}{I_{0}} - IR_{pv} \right )" /></a>, 
where

<a href="https://www.codecogs.com/eqnedit.php?latex=V_t&space;=&space;-(M&plus;R_{pv})\cdot&space;I_{sc}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?V_t&space;=&space;-(M&plus;R_{pv})\cdot&space;I_{sc}" title="V_t = -(M+R_{pv})\cdot I_{sc}" /></a>,

<a href="https://www.codecogs.com/eqnedit.php?latex=I_0&space;=&space;I_{sc}\cdot&space;e^{\frac{V_{oc}}{V_t}}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?I_0&space;=&space;I_{sc}\cdot&space;e^{\frac{V_{oc}}{V_t}}" title="I_0 = I_{sc}\cdot e^{\frac{V_{oc}}{V_t}}" /></a>,

<a href="https://www.codecogs.com/eqnedit.php?latex=R_{pv}&space;=&space;-M&space;\frac{I_{sc}}{I_{mpp}}&space;&plus;&space;\frac{V_{mpp}}{I_{mpp}}\left(1-\frac{I_{sc}}{I_{mpp}}&space;\right&space;)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?R_{pv}&space;=&space;-M&space;\frac{I_{sc}}{I_{mpp}}&space;&plus;&space;\frac{V_{mpp}}{I_{mpp}}\left(1-\frac{I_{sc}}{I_{mpp}}&space;\right&space;)" title="R_{pv} = -M \frac{I_{sc}}{I_{mpp}} + \frac{V_{mpp}}{I_{mpp}}\left(1-\frac{I_{sc}}{I_{mpp}} \right )" /></a>,

and with the numerically solved gradient
<a href="https://www.codecogs.com/eqnedit.php?latex=M&space;=&space;\frac{V_{oc}}{I_{sc}}\left(-5.411&space;\frac{V_{mpp}I_{mpp}}{I_{sc}V_{oc}}&plus;6.450\frac{V_{mpp}}{V_{oc}}&plus;3.417\frac{I_{mpp}}{I_{sc}}-4.422\right)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?M&space;=&space;\frac{V_{oc}}{I_{sc}}\left(-5.411&space;\frac{V_{mpp}I_{mpp}}{I_{sc}V_{oc}}&plus;6.450\frac{V_{mpp}}{V_{oc}}&plus;3.417\frac{I_{mpp}}{I_{sc}}-4.422\right)" title="M = \frac{V_{oc}}{I_{sc}}\left(-5.411 \frac{V_{mpp}I_{mpp}}{I_{sc}V_{oc}}+6.450\frac{V_{mpp}}{V_{oc}}+3.417\frac{I_{mpp}}{I_{sc}}-4.422\right)" /></a>.

This approximation does not consider the temperature coefficients. I'll look into it later, for now it's fine if we can just plot the V/I graph depending on the given four datasheet parameters.

@jurra You'd suggest using d3js? I will also have a look :) From my point of view it's not a priority, but it would be nice if we could make it work at least roughly in the next couple of weeks..

martinjaeger, 2019-06-19:
Additional possible tool for graphs: https://idyll-lang.org/

jurra, 2019-06-20:
It looks like they are using d3js behind the scenes.
This is an example of [idyll interactive content](https://idyll.pub/post/statistical-power-d9ff5d116b4c883d22a7888f/)
and this is [the source](https://github.com/mathisonian/statistical-power/tree/master/components).

martinjaeger, 2019-06-20:
Yeah, I had a closer look at idyll yesterday evening and it seems that idyll is an alternative to vuepress, as it also provides documentation + interactive things. But it doesn't seem suitable to be embedded into vuepress. So we'll better use vue.js directly, correct?

martinjaeger, 2019-06-20:
BTW, Michel added an additional wiki page for the selection of the suitable toolbox for graphs embedded to vue.js: https://github.com/LibreSolar/learn.libre.solar/wiki/Selection-Software-Environment-for-interactive-Toolbox

jurra, 2019-06-20:
I think the discussion is taking a different direction which its kind of a new issue based on the wiki note @mikelovic made. My suggestion is to go for one architecture that is extensible and useful for manythings. Vuepress is good because is based on the vuejs framework. It is a framework to develop consistent and easy to produce web applications. Vuepress is one application based on vuejs that was developed to facilitate project documentation. Because it is based on javascript, any javascript library for math, latex, imaging, etc can be integrated. My suggestion is to go for vuepress and extend it integrate it. Because it can be used for the learning resources, but also for repository and project documentation.

martinjaeger, 2019-06-20:
Maybe there was a misunderstanding. We want to keep using vuepress as the basis, but need some additional tools for the interactive graphs and also circuit diagrams etc.

One idea was that e.g. circuit diagrams are created with circuitikz and automatically rendered + deployed once they are pushed. This would allow anyone to change it without needing to use complicated software like inkscape, KiCad, etc. These circuits would then be embedded to the vuepress as images. Unfortunately, the vuepress latex plugin doesn't support custom latex packages like tikz, as far as I can see.

Regarding github wiki: If you quickly want to save some information which can be amended by everyone, they are easier to use than markdown files in the repository. And it's also more suitable than editing posts in issues. But once we are producing actual content, it will of course be in the git repository.

martinjaeger, 2019-08-23:
Interactive plotting successfully implemented using Chart.js plugin for VuePress. See [here](https://learn.libre.solar/b/load-switch/development/load_switch.html) for an example.

Documentation of solar panel chapter using the plots still in progress.

closed 2019-08-23 by martinjaeger